Chysn/O_C-HemisphereSuite #1: 1.3 Not Working

AWLIP, 2018-08-18:
I tried loading version 1.3 of the suite. Once I rebooted my OC, I was unable to get out of the load page. I pressed and held everything, nothing worked. The load bar completed it's cycle, but that's the only animation that you see. I reloaded version 1.2 and it's working like it was before. 
Since I'm leaving this note, might as well tack just one more on. I've noticed that the I can only get one quantizer to work at a time when using the dual quantizer app; the module will freeze and I'll have to reboot. 
I love the hemisphere suite, hope this feedback helps.

Chysn, 2018-08-18:
Hi! Did you compile in Arduino IDE, or install from the hexfile?

AWLIP, 2018-08-19:
I installed from the hex file.

Chysn, 2018-08-19:
OK. In that case, let's try clearing the EEPROM. Hold down the Up and Down buttons while starting your system. Keep them held down until the load bar gets to the bottom. You'll be asked to push the right encoder button to confirm.

This will clear app data on your O_C, but will not affect your calibration.

AWLIP, 2018-08-19:
Nice! That worked. And the dual quantizer issue that I was having before is
not occurring anymore.
I appreciate your help.

-Adam

Chysn, 2018-08-20:
Great! I'll close this issue, then.

closed 2018-08-20 by Chysn