Chysn/O_C-HemisphereSuite #6: LoRenz misbehaves

Chysn, 2018-08-28:
Found a repeatable Bug which freezes the Module (1.4.B)

-Left side:
LowerRenz

-Right side:
LoFi Tape (for example)

-Save State

-Power down

-Power Up

-Select App Selection on the right side

-Turn right encoder CCW <- Important!
when You get to the LowerRenz app it freezes!

Chysn, 2018-08-29:
Fixed in repo, not yet available in a hexfile

Chysn, 2018-09-01:
Released in 1.4C

closed 2018-09-01 by Chysn

benirose, 2021-12-19:
Sorry to comment on an old bug, but I'm experiencing a similar crashing issue in my build. Do you know what the root cause of this was?

Chysn, 2021-12-20:
> Sorry to comment on an old bug, but I'm experiencing a similar crashing issue in my build. Do you know what the root cause of this was?

Sorry, I don't remember that far back. You may need to check out the commit.

commit 029c2850 (2018-08-28): Fix #6