Chysn/O_C-HemisphereSuite #19: 1.6b (Flipped Screen) Trending Not working?

adrianfon9, 2018-11-06:
Hi Chysn, 

Trending does not detect anything but a "Steady" state in cv A B C or D throughout the range of sensitivity.

Thanks again for the quick response and sorry for spamming issues especially if they are all related to the same bug.

Chysn, 2018-11-06:
I'm unable to reproduce this issue in the flipped screen codebase. Can you verify that you're using the CV inputs and not the digital inputs?

adrianfon9, 2018-11-06:
Fixed on reboot. I apologize.

closed 2018-11-06 by adrianfon9

adrianfon9, 2018-11-07:
As an aside,

Is trending supposed to work with only 1 CV connected? I can only get either rising or falling based on if i plug cv into its corresponding input.

e.g. signal into cv 1; a:“rising” b:”falling” => trig A, no trig B
e.g. signal into cv 2; a:”rising” b: “falling” => no trig A, trig B

Adrian

Chysn, 2018-11-07:
No worries, thanks for helping me out!

Chysn, 2018-11-07:
Trending is now a dual applet as of 1.6. Both channels are independent. I’ll make sure to update the wiki!