Chysn/O_C-HemisphereSuite #23: FEATURE REQUEST

kurtjoy, 2018-11-13:
Hi Not sure if this is the right place to ask, but is it possible to adjust the minimum LFO speed so they are slower? Currently in the vector LFO it is 0.1Hz and in the skewed LFO is ~3sec? Can these be made even slower like the  quad LFO in the OG apps allowed? Cheers

Chysn, 2018-11-14:
Hi! Yes, this is a fine place for a feature request, and I always appreciate feedback.

For the Vector Oscillators, .10Hz is about the practical minimum for a single waveform. There are a number of reasons for this, but it's mostly because the processor doesn't have hardware floating point support, so I need to simulate floating points by shifting 32-bit integers. At a certain point, the resolution is too low, and the waveform no longer moves.

The big difference between the Vector Oscillator and classic O_C LFOs, why they can go slower, is that the classic O_C waveforms are simple. Vector Oscillators can have up to 12 segments, and each added segment increases the resolution needed to move within the waveform. It wasn't practical to have different ranges for two-segment waveforms and 12-segment waveforms, so I had to kind of thread the needle between flexibility in waveform design and minimum speed (on the other end of the equation, that's why the waveforms have a 12 segment max and not a 16 segment max).

The VectorLFO is designed to get around this limitation in one important respect, though. By default, the B/D output mixes two waveforms. Depending on the two speeds chosen, you can create a modulation source that doesn't repeat for a long time.

A worthwhile question, though!

kurtjoy, 2018-11-14:
Great, I figured that was the case with the vLFOs. Will definitely give the mixed mod speed trick a go!

closed 2018-11-14 by kurtjoy