Chysn/O_C-HemisphereSuite #31: Clock div app expects bipolar CV for div/mult, doesn’t work for Buchla CV

mathomas, 2019-03-05:
The clock divider/multiplier app accepts bipolar CV to adjust division/multiplication.  In the Buchla world, all CV is unipolar.  Thus it is impossible to reach the multiplication side of the app.

In a Buchla unipolar, 0-10v CV, world, it might make sense to have 0-4.99v represent multiplication and 5-10v represent division (essentially offsetting the input expectations).

Chysn, 2019-03-05:
Good call, thanks. I'm working on this now. Since the non-Buchla O_C has a non-symmetrical range (-3V to 5V), the CV range is -2.5V to 0V for multiplication and 0V to 2.5V for division. So for Buchla, the operative range will probably wind up being 2.5V to 5V and 5V to 7.5V. There will be a plus-or-minus-.4V detent centered around 5V to allow for manual (non CV) settings.

mathomas, 2019-03-06:
Sweet!  I assume it will clip at 7.5v if given more.  Not a problem, just want to test my guess.  The newer 2OC 1.5 appears to have attenuators on the CV ins, so that will be all the better.

Chysn, 2019-03-06:
Beyond 7.5V, it'll just max out the clock division value at 8.

Chysn, 2020-11-21:
Closing issue, as no further Buchla work is planned.

commit 2b7b2319 (2019-03-20): Fix #31: Buchla unipolar CV for div/mult
closed 2020-11-21 by Chysn