Chysn/O_C-HemisphereSuite #35: Possible to add Midi Clock to MidiOut?

CFenning, 2019-03-23:
Would it be possible to add "Midi Clock" to the Midi-In applet? Would be really awesome to have this functionality outside of CaptainMidi. 

Thanks,
Charlie

Chysn, 2019-03-24:
I think it's a great idea. I fear that there might not be enough memory left to do it. If this can't be done, it's only because I couldn't fit it in.

CFenning, 2019-03-24:
Great! Look for it in a future update? Or…? I’m new to the Hemisphere firmware.

CFenning, 2019-04-02:
> I think it's a great idea. I fear that there might not be enough memory left to do it. If this can't be done, it's only because I couldn't fit it in.

Any update on this?

kurtjoy, 2019-08-13:
Id like the reverse...Midi clock out in the "midi out" app

Chysn, 2019-08-13:
> Id like the reverse...Midi clock out in the "midi out" app

The Teensy 3.2 MIDI library does not have a way to construct a MIDI clock out message

Chysn, 2019-12-21:
MIDI Clock is added to MIDI In as of Hemisphere Suite 1.8, for release on 12/21/2019

closed 2019-12-21 by Chysn