Chysn/O_C-HemisphereSuite #36: ShiftGate: Channel 2 goes dark?

clayton-grey, 2019-03-30:
I was playing with ShiftGate this afternoon, and at one point I switched channel one from Gate to Trigger - or back? - I can't recall offhand, and channel 2 seemed to become stuck such that regardless of inputs the gate was seen as low so the pattern cleared out and no new outputs were generated. I tried reconnecting all cables with no luck. Resetting fixed, but it was in the middle of a practice set, which was unnerving.

Chysn, 2019-03-30:
Thanks for your report.

When you say "resetting fixed," are you talking to power-cycling your system, or something else?

Are you able to provide a procedure for reproducing this behavior?

Chysn, 2019-03-30:
I was unable to reproduce this while using ShiftGate. I suspect a transient hardware problem. I'll leave this open for a week or so, in case you reproduce it. If so, answer the following questions:

(1) Was ShiftGate in the left or right hemisphere?
(2) What was in the opposite hemisphere?
(3) When the condition begins, are you able to change to another applet?
(4) If so, does that applet have the same condition (that is, does its CV In 2 fail to function)?

clayton-grey, 2019-03-31:
You're probably right on hardware. but I'll be working with it a lot right now, so I'll keep you posted.

clayton-grey, 2019-04-05:
I've been using ShiftGate for at least 20 hours since this post with no issues, so this is probably safe to close. If it comes up again, I'll return!

closed 2019-04-05 by clayton-grey