Chysn/O_C-HemisphereSuite #37: Possible to add Quantermain back?

CFenning, 2019-04-02:
Would it be possible to add Quantermain back to the exclusive apps? Loosing the trigger functionality with DualQuant within Hemespheres is a deal breaker for me.

Chysn, 2019-04-02:
What do you mean by "losing the trigger functionality?" Is clocked mode not working in Dual Quantizer, or are you talking about something else?

CFenning, 2019-04-02:
Quantermain had the ability to select new note based on trigger/gate input. Now that I am running Hemisphere I have no ability to clock the Dual Quantizer - it just free quantizes as new voltages come in which is cool but not the most musical application.

CFenning, 2019-04-02:
Does the Clock Operation in the link you sent only work with steady clocks or will it accept unsteady/rhythmic clocks?

Chysn, 2019-04-02:
Clocked operation is basically a sample-and-hold. It grabs and quantizes the value at the time of the incoming clock, and holds the output at that value until the next clock, whenever that is.

CFenning, 2019-04-02:
Perfect! That should work for my needs. I will take a look. 

-C

closed 2019-04-02 by Chysn