Chysn/O_C-HemisphereSuite #38: Feature Inquiry: ShiftGate Normaling

clayton-grey, 2019-04-05:
Firstly, since discovering ShiftGate, I love it. It's **tremendously** useful! 

Is it possible to internally normal first channels output to the second channel's input in code? I'm doing this with stack cables now, and I'd love to save some cables and clutter. A pattern based on the other pattern is really useful to me, so mostly I'm curious if it's possible with the code and hardware (even if it's not something that goes into general release)?

Chysn, 2019-04-05:
It's possible, of course. Personally, I don't think that using a patch cable is enough of a hardship to introduce a new interface element, but that's why this is an open-source project.

closed 2019-04-05 by Chysn

Chysn, 2019-04-05:
I notice that you've contributed on development projects, so you can probably handle this. But if you'd like to contact me at my email address (from my GitHub profile), I'd be happy to give you any necessary guidance.