Chysn/O_C-HemisphereSuite #41: App Concept: Annular Fission (Quad Euclidian Circles)

clayton-grey, 2019-05-30:
I _may_ try to tackle this myself eventually, but I wanted to share this here - no expectations obviously (if there's a better place for this, please direct me). 

I understand the HEM file structure, but the APP's are more complex, and I'm not quite sure I understand well enough to adapt Annular Fusion into an APP version. Anywho, here is the notion:

[image]

Arrow above the numbers bounces 1 pixel up and down and indicates the circle the knobs are currently set to modify.

Knob 1, Fill
Knob 2, Length

Up Arrow, Last circle
Down Arrow, Next circle

Knob Button 1, Switch Control to rotation / fill
Knob Button 2, Maybe manual reset? Maybe toggle CV between reset and rotation?

Trigger in to clock the respective circle
Option to have the CV in be rotation or reset (not shown in UI mock)

Beyond this, the app behaves like Annular Fusion. The mockup uses rings that are sized to the smaller ring for Annular Fusion. I have trouble reading Annular Fusion as is, and I think having a quad version would help in this respect and otherwise seems like a compelling app.

Chysn, 2019-05-31:
Hey,

Yeah, that looks like it could be a fun project! If you're interested in
doing an O_C full app, that's what my blog focuses on:

http://feeds.feedburner.com/OrnamentAndCrimebutMostlyCrime

Regards,
Jason

Chysn, 2019-05-31:
Wait, that's the RSS link. Here's the actual blog:
http://butmostlycrime.blogspot.com/

clayton-grey, 2019-06-01:
And thanks for providing that APP_BoilerPlate. That should make this kind of adaptation much easier to digest!

Chysn, 2019-10-06:
Since this doesn't fit into any of the existing categories (bug, feature request, etc.), I'm going to close this issue. But feel free to contact me if you need any guidance; and if you do the app, I'd love to see it!

closed 2019-10-06 by Chysn

clayton-grey, 2019-10-07:
It's still in the maybe category. I'm debating about my approach to managing this problem with multiple potential solutions, but I still think this would be a useful application. Will definitely share if I do it!