Chysn/O_C-HemisphereSuite #42: Shift Register: Add gate out?

simonvanderveldt, 2019-06-02:
First of all: Thanks for the awesome firmware!

I was wondering if it would be possible to add a gate out, similar to the gate out supported by the Turing Machine/LFSR options on the original Ornament and Crime firmware

It's the  `> LFSR TRIG` setting.

From https://ornament-and-cri.me/user-manual-v1_3/#available-settings-per-channel-1
> aux channel output (when in LFSR mode): echo, lsb, chng: clock through, track LSB, and output on note-change

Chysn, 2019-06-02:
Hi,

Thanks for your input!

I’m not sure I understand the request, though. I’ve never used the original O_C firmware, and I don’t know what FSLR means.

However, both outputs of the ShiftGate applet are already used, so I don’t think there’s room to add anything. If you’re talking about adding an additional output mode, please explain the request.

Regards,
--jason

simonvanderveldt, 2019-06-10:
No problem :)

What the setting in the original O&C firmware (and on the pulses expander of the Turing machine module) does is generate a gate when the LSB flips or when the note changes or at the same rate as the clock/trigger (effectively acting as a mult/pass-through).

This allow one to use the Turing machines on the original O&C firmware to both generate pitch/melody and gate patterns, which is very useful/nice.

> However, both outputs of the ShiftGate applet are already used, so I don’t think there’s room to add anything. If you’re talking about adding an additional output mode, please explain the request.

Yeah, I noticed that. Personally I'd much prefer a gate over a second pitch output, but there are probably people who prefer a second pitch output ;)

Would it maybe be possible to make it a toggleable option? Or otherwise maybe a separate app or a build/compile time option?

Chysn, 2019-06-10:
Okay, I understand what you're saying. The Shift Register applet is designed to be a pretty faithful copy of the original Turing Machine. Check out Enigma, Jr. for assignable outs with gates and triggers.

closed 2019-12-21 by Chysn