Chysn/O_C-HemisphereSuite #44: HEM_Tuner bug in FLIP_180 mode

Afturmath, 2019-10-19:
When compiling with FLIP_180, the tuner warning/help messages indicate it should be run in the right hemisphere. The hardware constraint forces it to use digital 4 input which is the left hemisphere in flipped mode. Here is what I did to fix it, although this could probably be cleaned up.

diff of `HEM_Tuner.ino` (old to new):
```
74,75c74,86
<         if (hemisphere == 1) DrawTuner();
<         else DrawWarning();
---
>         if (hemisphere == 1) {
> #ifdef FLIP_180
>             DrawWarning();
> #else
>             DrawTuner();
> #endif
>         } else { 
> #ifdef FLIP_180
>             DrawTuner();
> #else
>             DrawWarning();
> #endif         
>         }
100c111,114
<             help[HEMISPHERE_HELP_DIGITALS] = "1=Input";
---
>             help[HEMISPHERE_HELP_DIGITALS] = "Due to hardware";
>             help[HEMISPHERE_HELP_CVS]      = "constraints, the";
>             help[HEMISPHERE_HELP_OUTS]     = "Tuner must run in";
>             help[HEMISPHERE_HELP_ENCODER]  = "left hemisphere";
103d116
< #endif
106a120
> #endif
107a122,127
> #ifdef FLIP_180
>             help[HEMISPHERE_HELP_DIGITALS] = "1=Input";
>             help[HEMISPHERE_HELP_CVS]      = "";
>             help[HEMISPHERE_HELP_OUTS]     = "";
>             help[HEMISPHERE_HELP_ENCODER]  = "A4 Hz P=Reset";
> #else
111,113d130
< #ifdef FLIP_180
<             help[HEMISPHERE_HELP_ENCODER]  = "left hemisphere";
< #else
```
Updated `View()` function:
```
    void View() {
        gfxHeader(applet_name());
        if (hemisphere == 1) {
#ifdef FLIP_180
            DrawWarning();
#else
            DrawTuner();
#endif
        } else { 
#ifdef FLIP_180
            DrawTuner();
#else
            DrawWarning();
#endif         
        }
    }
```
Updated `SetHelp()` function:
```
    void SetHelp() {
        if (hemisphere == 1) {
#ifdef FLIP_180
            help[HEMISPHERE_HELP_DIGITALS] = "Due to hardware";
            help[HEMISPHERE_HELP_CVS]      = "constraints, the";
            help[HEMISPHERE_HELP_OUTS]     = "Tuner must run in";
            help[HEMISPHERE_HELP_ENCODER]  = "left hemisphere";
#else
            help[HEMISPHERE_HELP_DIGITALS] = "2=Input";
            help[HEMISPHERE_HELP_CVS]      = "";
            help[HEMISPHERE_HELP_OUTS]     = "";
            help[HEMISPHERE_HELP_ENCODER]  = "A4 Hz P=Reset";
#endif
        } else {
#ifdef FLIP_180
            help[HEMISPHERE_HELP_DIGITALS] = "1=Input";
            help[HEMISPHERE_HELP_CVS]      = "";
            help[HEMISPHERE_HELP_OUTS]     = "";
            help[HEMISPHERE_HELP_ENCODER]  = "A4 Hz P=Reset";
#else
            help[HEMISPHERE_HELP_DIGITALS] = "Due to hardware";
            help[HEMISPHERE_HELP_CVS]      = "constraints, the";
            help[HEMISPHERE_HELP_OUTS]     = "Tuner must run in";
            help[HEMISPHERE_HELP_ENCODER]  = "right hemisphere";
#endif
        }
    }
```

Chysn, 2019-10-19:
Thank you for the report. Please note that this issue was addressed in Hemisphere Suite 1.6 [Release Notes](https://github.com/Chysn/O_C-HemisphereSuite/releases/tag/HSv1.6C) If you update to the latest version (1.7B), updates very similar to the ones you suggested will already be in place. https://github.com/Chysn/O_C-HemisphereSuite/releases/tag/HSv1.7B

Afturmath, 2019-10-20:
This bug affected 1.7B using the precompiled FLIP binary as well as the 1.8RC code in the current master. The above patches were necessary on the master branch downloaded October 19th.

Chysn, 2019-10-20:
Okay, thank you. Could you please pull the HEAD for 1.8RC and try it again?

Chysn, 2019-12-21:
No follow-up from reporter; assuming it's fixed in 1.8RC

commit bc0430c9 (2019-10-20): Fix #44
closed 2019-12-21 by Chysn