Chysn/O_C-HemisphereSuite #46: Support for Plum Audio 1uO_c4Robots?

versipellis, 2019-12-27:
Looks like if you try installing Hemisphere on Plum Audio's 1uO_c4Robots, your outputs are a mangled mess of voltages despite what the screen tells you. I'm under the impression it's a result of the VOR variable voltage outs that's been made a feature of it. Any chance we could get official mainline support for this? ISTR there being a beta firmware of Hemisphere that supports it that are pre-installed on the Pulplogic 1uO_c's.

Chysn, 2019-12-28:
There is a branch for VOR output (ocplus), and it is up to date with the Hemisphere Suite 1.8 feature set.

https://github.com/Chysn/O_C-HemisphereSuite/tree/ocplus

versipellis, 2019-12-28:
Thanks. Will this ever become part of the mainline release?

Chysn, 2019-12-28:
No. My preference would be for commercial vendors to maintain their own repositories to support their hardware, and merge in new feature updates themselves. Otherwise the master branch will become a maze of #ifdefs.

indigogo, 2020-03-17:
Is there any chance we could get a firmware image file (.hex) for this VOR branch? I've compiled a lot of code in my life but having a hell of a time trying to get this compiled (>_<)". It would be a wonderful gift to all your adoring fans (^_^)".

Chysn, 2020-03-17:
Hello, @indigogo. I don't have a complete codebase for Plum Audio products in this repository. I gave Plum Audio some early guidance, and the ocplus branch represents only a start. There are updates that they made afterwards that are not available to me. So, please contact Shay at Plum Audio, and he should be able to provide the latest hex file for the Plum Audio products! :)

indigogo, 2020-03-18:
Apologies, I did not look hard enough elsewhere before bothering you. It looks like they recently linked a hex from their site:
http://diy.plum-audio.com/projects/Firmwares/o_c_REV_HS_FixedCalibration%2B3Ranges.hex

Thank you for everything!

closed 2022-03-21 by Chysn