Chysn/O_C-HemisphereSuite #47: New Applet: Normalizer

wraybowling, 2019-12-30:
Hi. I'm just starting with Hemisphere and really enjoying it. I'd like to write a new applet that would take an input CV and allow it to be mapped to a new range. This would be handy for things like taking a clock from a pocket operator which is I think 1V and amplifying it up to 5V, or scale/offsetting LFOs, etc. There are plenty of analog scale/offset modules which can do this job without needing to involve an O_c but I've yet to see one that works like a map() function where the input minmax is mapped to an output minmax. If there's already an applet that does the job just as easily, feedback to stop trying to reinvent the wheel is welcome, otherwise some guidance on what I could read or who to speak to in order to start development is what i'm after. Also feedback on what else this applet would need to feel useful. My thoughts so far is that it should have a few different modes accessible by an encoder long-press
- map(a,b,c,d) style normalization
- rolling average / auto normalization / limiter
- scale + offset style normalization

wraybowling, 2019-12-30:
... I finally decide to post this when Attenuator/Offset was added probably the same day I had the idea with ca48991198c3add6c125f20199a82fd859ac1fbb. I think I see now that my idea should be broken up into separate applets, and i could simply make a PR if necessary to make sure that Attenuate/Offset is actually a Scale/Offset that allows for upscaling a signal, not just downscaling.

Chysn, 2022-03-21:
I'm closing all outstanding issues with the "enhancement" label at this time, as no additional enhancements are planned.

closed 2022-03-21 by Chysn