Chysn/O_C-HemisphereSuite #48: Enhancement: also invert when AttenOff <0%

Airell, 2020-01-13:
If the AttenOff goes below 0% (from 0 to -100%, one could also invert the CV signal.

Chysn, 2020-01-13:
Indeed, and I did consider that. The reason I didn't is that the module's output is asymmetrical, so the negative voltage would stop at -3V. +4V inverted would be -3V, as would +5V. It just seemed like a useless kind of inversion to me, one that would only annoy people.

Airell, 2020-01-14:
Aha. Thanks for the explanation.
I can understand why that would be unwanted. Personally I would not mind if this is a known limitation. At least you can invert it up to -3v ("60%-ish"). A function like a total inversion is not feasible this way.
On the other hand, this limitation is already a bit in there. You can choose to offset -5v so this will hit this limitation as well for signals below +2v.

Chysn, 2020-01-14:
Yeah, that's true. It can't be escaped. It's the most annoying thing about the module. I'll probably just go ahead with that update!

wraybowling, 2020-05-22:
I agree it'd be useful even if the range bottoms out at -3. It might actually visually teach a less familiar person about that voltage range limitation. Accept it! Embrace it! :)

giacecco, 2020-12-30:
@Chysn I've bumped into this matter by chance this morning testing AttenOff. I'm a newbie so forgive me if what I write is not correct or not expressed with the correct wording. Thanks to this GitHub issue I've learned about the Teensy input / output voltage limitation, I had missed it until today!

On the wiki [here](https://github.com/Chysn/O_C-HemisphereSuite/wiki/AttenOff) you write that the *offset* voltage range is -3V to +5V, but, unless my English is off, it would be more correct to remind the reader that it is the *output* range that is limited to -3V to +5V, that is the sum of the input and the offset. 

In fact, for example, I can use AttenOff to add to a +3V input a -5V offset.

Also, the input is subject to the same limitation, and it would be useful to remind the reader.

Thanks a lot for your work!

Chysn, 2022-06-21:
Indeed, I've updated the documentation