Chysn/O_C-HemisphereSuite #49: Captain Midi crashing on fast midi input

madskjeldgaard, 2020-02-11:
Hello 
First of all thank you for the fantastic work you have done here - Hemisphere is a daunting piece of work and I really love it so much!

I experience an issue with Captain Midi crashing the Teensy on an otherwise perfectly functioning module. I have fairly fast CV signals going into all four CV inputs which are then fed to channels 1-4 as mod data. It works fine for some time but at some point, inevitebly it seems, it crashes the whole Teensy leading to glitches on the screen and the whole thing dead until reboot.

I am running Hemisphere v1.8 on a mini OC with a Teensy 3.2 overclocked to 120 mhz as per the OC instructions.

I hope this information is useful and otherwise let me know if you want more detail. I atttach here a picture of the dead Captain Midi.

Thanks

[image]

madskjeldgaard, 2020-02-11:
Update: This seems to happen only/mostly when pulling up the log display. Or at least that seems to trigger the crash when doing that.

madskjeldgaard, 2020-02-11:
aha, okay, even though it doesn't crash with 4 high frequenct/audio rate inputs the menu becomes sluggish and unresponsive

Chysn, 2020-02-11:
Hello, and thanks for the report!

Did you install using the hex file from the Release page in GitHub, or via the Arduino IDE?

madskjeldgaard, 2020-02-11:
I am not 100% certain but as far as I remember I did install it using
the hex file.

Chysn, 2020-02-11:
I’m asking because the IDE version is really important for stability reasons. And you mentioned overclocking, which wouldn’t be an issue with the hex file.

I’d say, re-install using the hex file and see if the issue goes away.

madskjeldgaard, 2020-02-11:
> I’m asking because the IDE version is really important for stability reasons. And you mentioned overclocking, which wouldn’t be an issue with the hex file.
> 
> I’d say, re-install using the hex file and see if the issue goes away.
> 
> https://github.com/Chysn/O_C-HemisphereSuite/releases/

Okay thanks, I will try that and report back here with the results.

madskjeldgaard, 2020-02-11:
So I tried that now. I realised I was using arduino 1.8.11 and teensyduino 1.5 and I downgraded to the supported versions 1.8.1/1.35 and tried both compiling myself via the IDE and uploading the hex file but all of it leads to the same issue with Captain Midi crashing unfortunately. 

I also tried resetting the eeprom but that did not help either. 

As I said above, the rest of Hemisphere works as expected, it's just Captain Midi. 

I should add that it gets progressively sluggish the more cv inputs I patch with these signals, it starts to slow down from the third cv input for me and then after a while with the fourth plugged in it crashes.

Chysn, 2020-02-11:
Ok, thanks for checking that!

I’ll look into this today. It’s four audio rate CV, assigned to four continuous controller messages on different MIDI channels?

madskjeldgaard, 2020-02-11:
Thanks for having a look! Yes, Midi Assign is set to Mod for all four channels, the channels are 1-4. Everything else is set to off

Chysn, 2020-02-12:
Fixed in 1.8A. This looks like a bug from the O_C framework, in which some integers would not print properly, and caused a buffer overwrite of screen memory, which usually crashes the module. Fixed by moving the log values display a bit out of the way.

closed 2020-02-12 by Chysn

madskjeldgaard, 2020-02-12:
Well done ! I will test out the fix here

madskjeldgaard, 2020-02-12:
I recompiled from the master branch including the change and it all seems to be working perfectly now. Thanks a lot for fixing this!

Chysn, 2020-02-13:
Sure, thanks for the report!

KittenVillage, 2020-03-10:
I think I just hit this bug. I'll try to figure out how to compile on my own, I have a feeling I'll run into it again with the setup I'm trying out. (4 channels in, set to note.)

[image]

filefolder, 2020-04-13:
i'm also getting this bug FWIW. using the hex file, haven't tried compiling. every time i accidentally pull up the routing info it crashes almost immediately with busy midi traffic. 

also i don't know how to save the config (pushing left encoder for 2 sec doesn't do anything) but that's a different problem

Chysn, 2020-04-13:
> i'm also getting this bug FWIW. using the hex file, haven't tried compiling. every time i accidentally pull up the routing info it crashes almost immediately with busy midi traffic.
> 
> is there a way to download the updated 1.8A hex somewhere?

Not yet, but it will be available this week.

filefolder, 2020-04-21:
heyyy at the risk of gift-mouth-horse-looking could i say <bump> or can anyone just post their 1.8A hex in this thread? this bug is absolutely debilitating for me. thanks in advance

Chysn, 2020-04-21:
> heyyy at the risk of gift-mouth-horse-looking could i say or can anyone

Sure, this is all set. It's in the Assets section of the 1.8 release:
https://github.com/Chysn/O_C-HemisphereSuite/releases

Regards,
Jason

KittenVillage, 2020-05-06:
I compiled with a flipped screen and got this error again. Not sure if it's because I pulled the source off the 'releases' page or not. I'll give it go with the source pulled down from the main page and report back in a bit.

KittenVillage, 2020-05-06:
Is working fine with flipped screen. The arduino IDE does some caching or something? I finally had to delete the old directory and start fresh to make sure I had the latest version.

Chysn, 2020-05-07:
I know that the Arduino IDE does some kind of caching, but I don’t really know how it behaves. I’m glad you got it working.

commit 46f231d8 (2020-02-12): Fix #49