Chysn/O_C-HemisphereSuite #52: Gated VCA app stops working when screen goes to sleep

synaptyx, 2020-02-19:
I have the Gated VCA running, but when the screen goes to sleep, the GVCA stops working, or at least is mostly muted. I am activating the GVCA with a trigger from Euclidean Circles V2 and the GVCA is managing noise sent to a T-Networks percussion module to create a clap/snare-like sound. It does the same if manually activating the screen-saver, or just letting the screen time-out on its own

Chysn, 2020-02-21:
I have not been able to reproduce this behavior. The Gated VCA works the same regardless of whether the screen is blanked or not.

Do you observe this with any other applets? What version of Hemisphere Suite are you using?

synaptyx, 2020-02-22:
Hello, thanks for your response. I'm on v1.8, having just flashed my o_C module a few weeks ago. Here is a video of the issue: https://youtu.be/NdCqm9qIDH0

Chysn, 2020-02-22:
Thank you for the video, that is helpful. This is an interesting situation, because the only thing that the screen blanking does is to avoid drawing the screen. The I/O function and the screen function are totally independent. The only difference between drawing the screen and not drawing the screen, from a functional standpoint, is the time it takes to draw the screen.

In general, I would say that using a trigger to gate the VCA is an unreliable way to gate the VCA. Triggers are usually super-short. I think the only reason that your Circles trigger is of sufficient length to make a sound is that, once the trigger is received, the module needs to spend some time generating the screen. Absent that "artificial" time extension, the VCA closes too quickly, and reduces the perceived length of the sound. When the screen is blanked, you're hearing the _real_ trigger length.

Recommended solution: Rather than sending the Circles trigger directly, you may need to convert it to a longer gate before sending it to Gated VCA.

I do not think that there is anything that needs to be done within the Hemisphere software, so I'll close this issue. However, feel free to respond if you have further questions.

Thanks!

closed 2020-02-22 by Chysn

synaptyx, 2020-02-22:
Agreed, more of a usage issue. Thanks for your time and attention, it's most appreciated.