Chysn/O_C-HemisphereSuite #53: BootsNCat - Kick

mannstrane, 2020-02-25:
Is there anyway to adjust the attack on the Kick so it doesn't click?  I get inconsistency with the Kick causing an audible click on output.  It needs the slightest adjustment to keep from clicking at higher gain settings on output.

Chysn, 2020-02-25:
No, the decay can be adjusted with the encoders, but not the attack.

--Jason

mannstrane, 2020-02-25:
Thanks for the reply.  Is it normal to get a clicking with kick or is it something with my unit?

wraybowling, 2020-02-25:
@mannstrane post a sound clip if possible? I can compare with mine when I get home in a couple of hours.

Chysn, 2020-02-25:
I guess it is a little on the clicky side.

mannstrane, 2020-02-25:
@wraybowling, I will post later tonight.  Would be comparable to cutting a waveform at the rise of a transient compared to cutting at the 0 level.  Makes me wonder if there is some sort of variable delay with the waveform.

mannstrane, 2020-02-26:
Here is a screenshot of the waveform:
https://imgur.com/a/xEWEjF2  
First transient is ok.  2nd transient has a spike along with some after that.  

Attaching wav file here:
https://we.tl/t-2K3j794894

mannstrane, 2020-02-27:
Has anyone had a chance to see if the results I am getting are software vs a hardware issue affecting my unit?

wraybowling, 2020-02-27:
I have listened to the file you uploaded. I have also been listening to [other kick drums on free sound](https://freesound.org/search/?q=kick+drum&f=&s=score+desc&advanced=0&g=1). I have not checked my unit yet but imagine that it's going to sound the same due to the fact that O_c isn't meant to really run at audio rates so some clickiness might be just sample rate getting weird. But it begged the question of what kind of kick would sound good at that lower rate. (i'm making assumptions based on the original O_c, not Hemisphere Suite). Might be able to come up with something that strikes a better balance so that the Kick has mostly lower frequencies and the Snare has mostly mid and upper frequencies.

Looking at the source for BootsNCat(s), both sounds are synthesized from a triangle wave and a sawtooth wave. The click is likely coming from when the envelope lines up with a sawtooth peak.

mannstrane, 2020-02-27:
@wraybowling 

Did you look at the image I posted above?  You can see an inconsistent spike in the waveform.  Would be good to know if the drift in envelope timing is software or hardware related.  Does everyone get the same result?

wraybowling, 2020-02-27:
I did. That's because the wave is always cycling regardless of the envelope. I don't know the code base well enough yet to know if I'm right in this suggestion... (@Chysn?) But I THINK eg[ch].Cycle(0); would be a hard sync and that would remedy the envelope and audio waves making inconsistent transients.

mannstrane, 2020-02-27:
Interesting.  Well I am curious to know if the o_C unit I am using is faulty or if this is a micro controller limitation.    I was really looking forward to using this firmware for a minimal system with dual function ability.  Adding a VCA and Envelope to the signal path after the o_C somewhat ruins the experience this firmware and hardware has to offer with Kick/Snare.

Chysn, 2020-02-28:
The vector oscillators are not free-running. The trigger to start the
vector oscillator also resets it to the beginning of its cycle. If there's
variation in the sound from one instance to the next, it's more likely that
it's due to variations in the *end* of the *previous* instance.

mannstrane, 2020-02-28:
@Chysn 
So then the record sample I provided previously is by design and identical for all firmware installs for the kick?  Is there a possible software remedy to have the kick fixed?

junedoom, 2020-03-01:
The envelope vector oscillators are reset but the audio rate triangle vector oscillator for the bass drum is not reset so the phase is inconsistent on the trigger.

Chysn, 2020-03-03:
> The envelope vector oscillators are reset but the audio rate triangle
vector oscillator for the bass drum is not reset so the phase is
inconsistent on the trigger.

Yes, confirmed. That would be the cause. I may try resetting the bass
oscillator on the trigger.

>

wraybowling, 2020-03-03:
That would remedy the issue! Bass drums really need that consistency. The waveform always starts out as mostly envelope and then has tone. For a first method I’d try syncing the tone. If that doesn’t work Maybe a second envelope added to the signal would give it that “kick.”

mannstrane, 2020-03-03:
Thanks again for looking into this everyone.  I really love this little app.  Looking forward to a possible update.

Chysn, 2022-03-21:
I'm closing all outstanding issues with the "enhancement" label at this time, as no additional enhancements are planned.

closed 2022-03-21 by Chysn