Chysn/O_C-HemisphereSuite #54: Unlinked "A"SR

versipellis, 2020-03-19:
Is it possible to run "A"SR in both hemispheres without being linked? I'd essentially like to be able to have two shifted inputs, and Hemisphere is the most efficient channels/hp there is if it's able to be run unlinked.

Chysn, 2020-03-19:
If it detects "A"SR running in both hemispheres, they will always be linked. The reason for this is that there was only enough memory available for a single ring buffer. So when two copies of "A"SR are run, the second one is just sort of an "expander" with additional outputs for the one buffer.

versipellis, 2020-03-19:
As a possible, as a suggestion in the future, to have two ring buffers with a shorter buffer length as a tradeoff?

Chysn, 2020-03-19:
Yes, that's a good idea.

versipellis, 2020-03-21:
Should I open that as a new issue for feature request?

Chysn, 2020-03-21:
I'll re-open this one and add a the Enhancement label. But I have to be honest, I don't plan any further development of Hemisphere Suite at this time.

Chysn, 2022-03-21:
I'm closing all outstanding issues with the "enhancement" label at this time, as no additional enhancements are planned.

closed 2022-03-21 by Chysn