Chysn/O_C-HemisphereSuite #57: Hemispheres VOR

CorradoMorgana, 2020-05-07:
Getting some strange results with Hemispheres. Carpeggio mainly, sometimes Anuler

Sometimes CV works, sometimes trigger works, sometimes both

I think a firmware issue as the CV lights function fine and all seems good with Original VOR U-OC firmware.

Thanks in advance 

*Update.

I think I have found a solution.

The CV knobs really have to be at ZERO and Unipolar

*Update 2

Nope that ain't it...

Chysn, 2020-05-07:
Could you please provide a procedure to reproduce?

CorradoMorgana, 2020-05-07:
Hi There,

Not sure if I can...but I'll try as it seems sporadic.

I have no issues with U_OC for VOR  (Plum audio, Thonk Kit)
No issues (so far) with Hemispheres VOR other than with Anuler and
Carpeggio.

On boot up, I select the app as normal
Anuler and Carpeggio both have CV and Digital Trigger control

In Carpeggio I can SOMETIMES use CV 1 and 2 to move between the note grid
(cartesian mode)
In Carpeggio I can SOMETIMES use  Digital Trigger 1 to linear sequence
Sometimes I can use Both Linear and Cartesian via trigger and CV
Red/Green CV lights operate as expected in both Bi and Uni mode

On one occasion turning the CV knobs to zero and the Bi/uni switch to Uni
allowed me to use Linear mode. Turning the CV knobs Cartesian x,y scanned
the grid as normal
I switched app or re-booted, this stopped working and either Cartesian or
Linear only worked (can't remember which)

Similar issue with Anular.
Trigger works, but CV to rotate rings sometimes fails

Calibration issue? CV deadzone?

Hope that helps.

Great set of apps btw

Thanks

Corrado

Chysn, 2020-05-07:
Huh. Try re-seating the Teensy. I have trouble with my inputs if my USB cable hangs up on a neighboring module and knocks the Teensy partially out of the socket.

CorradoMorgana, 2020-05-08:
Hi There,

Checked...well seated.

My UoC is the Plum Audio version with VOR and USB header on the front
panel. Only time I need to fiddle with the Teensy is to press the program
button.

Aah, well, thanks anyhow.

Will keep testing. It's all still fun

Best

C

Chysn, 2020-11-21:
Closing this, because I plan no further work with VOR

closed 2020-11-21 by Chysn