Chysn/O_C-HemisphereSuite #58: Device fails to boot after installing 1.8

chrisfromwork, 2020-05-11:
After using Teensy.exe to install the 1.8 hex file, the device shows the expected software on the screen upon reboot.

After unplugging from the computer and turning off/on case power, the device no longer boots.

A workaround for anyone that hits this is to go into backup/restore and backup the current software. It seems like something may be wrong when no backup has been performed.

wraybowling, 2020-05-16:
I might be experiencing the same thing. I just flashed 46f231d8baaa9e10957efbb65faf01ebc253ca9b and got a black screen. Will try investigating..

wraybowling, 2020-06-04:
@chrisfromwork Did you try flashing the 1.8 or 1.8A hex files that are linked in Releases? those work. I just figured out something must be wrong with my setup as even old versions are building hex files that produce nothing but a black screen.

closed 2020-11-21 by Chysn