Chysn/O_C-HemisphereSuite #59: Metronome in Forward clock mode no longer has start/stop

wraybowling, 2020-05-16:
In releases prior to 1.8, the user could
- open metronome
- select play/pause
- Press the encoder to start/stop the clock

Starting with 1.8, metronome only is a visualization for the internal clock but it cannot start/stop. If the user selects forward clock mode (long press down arrow, select forward), then the Metronome no longer functions because the user cannot start/stop. Since it cannot start/stop, metronome cannot serve as a clock to other modules without influencing the other hemisphere via internal clock mode.

Chysn, 2020-05-17:
Metronome was introduced in 1.6 as a way to internally clock both hemispheres, and to send clock to other modules. While this internal clock is running, clock forwarding did not function. The 1.8 change was a way to make the clock globally-available instead of requiring the musician to scroll to the Metronome to make changes.

So now, instead of having to scroll over to Metronome to start and stop, you long-hold and then release the left button. The functionality is identical, but is now more easily accessible.

wraybowling, 2020-05-18:
I'm familiar with the new functionality. I'm maybe not clear on what the forward clock mode actually is. I'm trying to reconcile functionality while you're in forward clock mode because it's still an option.

closed 2020-11-21 by Chysn