Chysn/O_C-HemisphereSuite #62: Upgrade from Arduino IDE 1.8.1 to 1.8.12

wraybowling, 2020-06-06:
I asked on facebook how to get O_C HS to build a working hex file and was told I have to use Arduino IDE 1.8.1 Seems a bit old and like we could upgrade to 1.8.12.

Note I'm also trying to address this sort of incongruency stuff with PR #61

Chysn, 2020-06-06:
Yes, use 1.8.1 to build Hemisphere Suite. Ornament and Crime has the same requirement; after 1.8.1, Arduino IDE started using gcc 5.4, which does not support Teensy 3.2 reliably.

closed 2020-06-06 by Chysn

wraybowling, 2020-06-07:
Sorry to drag this through the mud, but by that reasoning, does that mean that HS should be built with Teensyduino v1.35? (I see that's the case on O_C as well, just confirming!) If so I will add that info to the wiki. I think this will be an important note as these versions will become legacy software and fall out of support—especially on MacOS where Teensyduino.

Chysn, 2020-06-07:
Yeah, Teensyduino 1.35 is correct.

benirose, 2021-11-10:
Sorry to comment on a closed ticket. I can't seem to install v1.35 of Teensyduino due to either it or the installer being a 32-bit application. Is there any alternative at this point or do I just need to find an old box to dev on?