Chysn/O_C-HemisphereSuite #64: Midi In - Change Clock PPQ

CFenning, 2020-06-22:
Hi Jason,
I have been using the Clock In that you added to Midi-In after I posted about it months ago - You're the man, thank you!

Right now my module puts out 2PPQ which works but is a little less than optimal. Would it be possible to add the ability to change the PPQ to a higher-resolution?

Thanks!
Charlie

CFenning, 2020-06-23:
Would it be as simple as changing Line-23? 24PPQN /12 = the slow 8th-note clock I am currently getting.

"#define HEM_MIDI_CLOCK_DIVISOR 12"

Can I just change 12 to 1?

Chysn, 2020-06-24:
Yeah, that would work. You'll get 24 pulses per quarter note with incoming clock.

CFenning, 2020-06-26:
I am having issues compiling the software in Teensyduino - my first time playing with the code.

When I open up o_c_REV, its saying "usbMIDI was not declared in this scope" 

Any help would be appreciated.

Chysn, 2020-06-26:
Can you please email me at jjustian@gmail.com, and I'll help you out. The Issues system isn't the best place for this kind of thing. Thanks!

closed 2020-06-26 by Chysn