Chysn/O_C-HemisphereSuite #65: Possible bug: clock div encoder problem

jull-taragan, 2020-06-22:
Clock div doesn't respond to encoder turn. Using 1.8 flipped.
Only works on the bottom clock and only on the right panel. Other apps respond to encoder turns and I can browse menus and stuff.
Sorry if this is not properly formed.

Chysn, 2020-06-23:
Almost always, this symptom indicates a calibration issue. The clock division can be controlled by incoming CV. Since the module has no way to detect whether a jack is plugged in, there's a software "detent" around 0V, and the ADC voltage must be within that detent to allow panel control. If your ADCs aren't calibrated properly, incoming 0V may exceed the detent voltage threshold (in one direction or the other), causing the voltage control to override the panel control.

Flipped modules are particularly vulnerable to this, because builders usually calibrate for regular orientation, and a flipped module needs to be completely recalibrated because the inputs and outputs are in reverse order, and the software thinks what is now In 1 is still In 4.

closed 2020-06-23 by Chysn

jull-taragan, 2020-06-24:
OH WOW! Thanks a lot friend,
Sorry for noobing :) Amazing advice.
Thanks again,
Jull.
*תודה רבה,*

*ג'ול טרגן - 054-6133102*

jull-taragan, 2020-06-24:
Worked like a charm.
Thanks once again!
Jull
*תודה רבה,*

*ג'ול טרגן - 054-6133102*