Chysn/O_C-HemisphereSuite #70: VOR bug with AttenOff

esmooov, 2020-08-12:
Testing with the new version, I'm noticing that there's a bug in how the AttenOff app has been adapted for VOR support. For instances when I set the outputs to -5 -> 5 and try to use the AttenOff hemisphere function, the module refuses to output a voltage above 0 volts. When I set the outputs to -3 -> 7, AttenOff maxes out at 2v. I hope this is the right place for this feedback; it's a little unclear to me who owns the VOR features.

MuTron1, 2020-08-17:
I am having the same issue with other applets on the Plum Audio OCP

**Voltage**
0 to +10V range: +5v in Voltage gives me +5v output. Output corresponds to voltage until I go below zero. At -1v selected on Voltages, I still get zero volts. In this range, -5v selected should give an output of 0v, and +5v should give an output of 10v, to give the full 10v range

-3 to +7v range: +5v in Voltage gives an output of +2v. 0v in Voltage gives an output of -3v. Anything lower in Voltage gives -3v still

-5v to + 5v range: +5v in Voltage gives an output of 0v. 0v in Voltage gives an output of -5v. all negative voltages still read at -5v output.

**Vector LFO**

At the 0 to 10v range: All LFOs peak at about 3v but clip at 0v. The bipolar sine, for example, spends 50% of it's cycle flat at 0v

moving the ranges to asymmetrical and bipolar as the same waveform. The zero crossing point, rather than being at the mid point of the range, is at the lowest voltage of the range, clipping the rest of the waveform.

Similar issues are in Carpeggio, with the first few octaves being flat at 0v/-3v/-5v.

O_c standard firmware behaves as expected (Voltage, for example, gives the lowest voltage in the selected range, +1 Octave gives a volt higher, etc) so the hardware has no build errors. 

Seemingly an error in the VOR method, where the applets themselves are trying to output a voltage offset -5v from the output range selected.

Chysn, 2020-11-21:
I'm closing this, as I plan no further work with VOR

closed 2020-11-21 by Chysn