Chysn/O_C-HemisphereSuite #75: Explain buchla build flags in the wiki

madskjeldgaard, 2020-09-17:
Hi! Just (succesfully ---> https://vimeo.com/458966794) installed hemisphere on a cardOC on my Buchla and I got it working fairly well but I got confused by the fact that there are so many build flags available for buchla. Can you make a wiki note explaining them? Is only one needed? Is two? I know this may sound stupid but it's a bit confusing  (for me at least). Again, thanks for this amazing project!!

madskjeldgaard, 2020-09-17:
Say for example I was to compile for the cardOC. These are the flags available in OC_options.h:
``` 

/* ------------ uncomment for Buchla/non-octaval/quartertone support: -------------------------------  */
//#define BUCHLA_SUPPORT
/* ------------ uncomment for use with Northernlight cOC program card: ------------------------------  */
//#define BUCHLA_cOC
/* ------------ uncomment for use with Northernlight 4U / cOC or 2OC: -------------------------------  */
//#define BUCHLA_4U
```
The first is necessary I guess but then do I also use 2 and/or 3? It seems from the comments. 

Thanks for clearing it up :)

Chysn, 2020-09-17:
These are options carried over from Ornament and Crime, so I'm not that familiar with them myself. My best guess is that for a Northernlight cardOC, you'd use all three of them.

The only one I added was the one that specifies the left-hand or right-hand side of a 2OC, because some of the controls get flipped.

closed 2020-09-17 by Chysn