Chysn/O_C-HemisphereSuite #77: Carpeggio, display position/note value does not reflect the played and edited note

Airell, 2020-11-21:
The note and position that is displayed is not the note you hear and edit. The display and the shown note is one position further. When the unit starts, you hear the last note, but the first step is shown. That the last note is selected on start is understandable, because when you start the sequence, it advances to the first step.

E.g. when in 'Octave', the first note is C2, the second note is C3. When the sequence stops, and note 2 is shown on the display (C3), you hear note 1 (C2). Also when you edit this note, you don't edit the second note, but the first, which is confusing.

Chysn, 2020-11-21:
@Airell : Thank you so much for the bug report. This is fixed in the repo (production branch), and will be in the 1.8A release.

closed 2020-11-21 by Chysn

Airell, 2020-11-23:
I have tested today. It works great.
Much easier to manually step through the sequence now and change a note.
Thank you!

commit dc973e0a (2020-11-21): Fix #77