Chysn/O_C-HemisphereSuite #78: Enhancement/new feature: have (c)arpeggio react to 1v/oct on CV-in for root selection

Airell, 2020-12-03:
Hi, this is an idea of possibly reusing the carpeggio app for a more 'normal' new arpeggio 'app'. Maybe internally the carpeggio code could be reused, as only the CV inputs change function a bit. This might keep the extra code low, I don't know if this possible.

If the CV1 in (for X) could be changed to set the transpose, preferably 1v/oct (like root note), you could 'play' it without extra modules like precession adders. The CV2 (for y) might become something else... (direction, chord selection, arp length).

The many chords already in there are very nice for a unique 'arp'.

Thanks for consideration.
Airell.

no-trbl-2-u, 2021-04-02:
I really like this idea :) I think I saw something in a VCVRack-Hack video where they just use `log2(x)` in order to convert a CV signal to track 1V/Oct, so theoretically, this might be a pretty easy request. (I could be way off, though 😆 )

Chysn, 2022-03-21:
I'm closing all outstanding issues with the "enhancement" label at this time, as no additional enhancements are planned.

closed 2022-03-21 by Chysn