Chysn/O_C-HemisphereSuite #79: Compiling: Blank screen

mcleinn, 2021-01-05:
Dear community, 

I have been using Hemisphere Suite for quite a while, all versions, including the latest, install normally via HEX file. Working 100%.

Now I wanted to build the suite by myself. Unfortunately, I am getting a blank screen after reboot for all latest versions. I went back in history - all versions up to 1.5A compile and run normally (the latest O_C release without hemisphere compiles and runs, as well). But all versions beginning from 1.6 result in a blank screen.

Would you have any suspicions regarding possible reasons for this behavior? 

Thank you

mz2250, 2021-01-06:
Hi, you really have to use both Arduino ide and teensyduino version suggested in Ornament firmware upload procedure manual.

"method B: compiling the firmware
step 1): get the IDE + teensyduino add-on
if you don’t have it already, you need to install the Arduino IDE as well as the Teensyduino v1.35 add-on. Note that you MUST use specific versions of these! At the time of writing, that’s Arduino IDE 1.8.1 (available from https://www.arduino.cc/en/Main/OldSoftwareReleases - just scroll down and it’s in the table) and Teensyduino v1.35 ([Mac] [Windows] [Linux]), respectively; at this point, later versions of Teensyduino (which use gcc 5.4 instead of gcc 4.8) do not work, so please don’t use them (they will cause screen corruptions). Only use Teensyduino v1.35!"

You can download the teensyduino here

https://ornament-and-cri.me/firmware/

Chysn, 2021-01-06:
This is more of a support issue than a bug report, so it doesn't really belong here. I'm going to close the issue. Build with the versions specified in the documentation, and all will be fine.

closed 2021-01-06 by Chysn

mcleinn, 2021-01-06:
I saw that, but it is quite impractical to ask people who are doing all
kinds of modern Teensy and Arduino development to install IDE and libraries
which are several years old. In fact, compilation does work with modern
libraries, just with the small code optimization, which is largely
identical to - o2.

mz2250, 2021-01-06:
That was already discussed a lot, I saw it's something related to the first version of ornament and can't be changed, you can always have on your machine both the teensyduino 2 and 1.35 and both the newer arduino ide and 1.8.1, just rename the app to something like arduino_2 and it will work

mcleinn, 2021-01-06:
I see. But, as I wrote, it is actually compiling and running with the
latest version of IDE and libraries, I just needed to switch on "small
code" optimization. So I wonder if this rule, which could turn away people,
is still necessary. Are there any tests you can recommend to see if it
works at the regular frequency?

mz2250, 2021-01-06:
Here's a response from chysn itself on lines forum to same request.

https://llllllll.co/t/hemisphere-suite-support/21704/33

I think it's a better investigation asking in original o_C firmware git from mxmxmx since hemisphere it's an alternative Firmware.

But I'm not speaking on behalf of chysn, so maybe he can add more on that.