Chysn/O_C-HemisphereSuite #80: "ShiftReg" in 2nd hemisphere. Loop length stuck on "2"

CompositeCone, 2021-01-07:
Hi! on my version of hemisphere suite, specifically in the right hemisphere and using Shift Register, I cant control the loop length by twisting the encoders. It is just stuck on "2" steps. When I try and move the encoder it instantly just snaps back to "2" even with nothing patched in. Not too sure if this is a firmware bug or I am missing something. Video is attached.

https://user-images.githubusercontent.com/77107623/103910513-056e8380-5159-11eb-9127-5584a9013e08.mp4

Chysn, 2021-01-07:
This indicates poor CV input calibration. Calibrate the module and try again.

closed 2021-01-07 by Chysn

CompositeCone, 2021-01-07:
Thanks. the matter seemed to have sorted itself when trying it later in the day. Very strange but I will re calibrate. Thank you for the help!!