Chysn/O_C-HemisphereSuite #81: Feature request: Add 14 bit midi support to captain midi

madskjeldgaard, 2021-01-21:
Hi ! 
It would be awesome if captain midi had support for 14 bit midi which is supported by some software (like REAPER, SuperCollider, Max, PD, etc.)

I would imagine adding "MOD14" to the list of parameter types in the midi settings after MOD. 

14 BIT midi entails sending two midi cc messages at the same time which seems taxing on the hemisphere.

I have a local branch running with some changes I made to the firmware to add support for 14 bit midi. It could look something like this in the APP_MIDI.ino file

```
                // 14 bit cc
                if (out_fn == MIDI_OUT_MOD14 || out_fn >= MIDI_OUT_EXPRESSION) {
                    int cc = 1; // Modulation wheel
                    if (out_fn == MIDI_OUT_EXPRESSION) cc = 11;
                    if (out_fn == MIDI_OUT_PAN) cc = 10;
                    if (out_fn == MIDI_OUT_HOLD) cc = 64;
                    if (out_fn == MIDI_OUT_BREATH) cc = 2;
                    if (out_fn == MIDI_OUT_Y_AXIS) cc = 74;

                    int value = Proportion(In(ch), HSAPPLICATION_5V, 127);
                    if (cc == 64) value = (value >= 60) ? 127 : 0; // On or off for sustain pedal

					// 14 bit
                    usbMIDI.sendControlChange(cc, value & 0x7F, out_ch);
					usbMIDI.sendControlChange(cc + 32, value >> 7, out_ch);
                    UpdateLog(0, ch, 2, out_ch, cc, value);
                    indicator = 1;
                }

```

madskjeldgaard, 2021-01-21:
With 14 bit midi you get an almost CV-like resolution to the midi which is truly awesome :)

madskjeldgaard, 2021-01-21:
And also, I can make a PR for this if you are interested

Chysn, 2022-03-21:
I'm closing all outstanding issues with the "enhancement" label at this time, as no additional enhancements are planned.

closed 2022-03-21 by Chysn