Chysn/O_C-HemisphereSuite #86: Scale Duet problem (only first scale works)

texasheadache, 2021-04-22:
Hi - in the past i have used the scale duet to switch between two scales with ease. Recently i've found that only the first scale now works - not the second. This is true for both hemispheres. I am not sure how long this has been the case but now that i think of it I believe it has been this way for a while. I have updated to the newest firmware and have reset the EPROM settings. I am using a square wave to trigger the clock and a random source to change the notes. Any advice? Thank you for your time.

Chysn, 2021-04-22:
I'm unable to reproduce this issue with version 1.8A. Scale switching works as expected, by switching to Scale 2 when the gate at Digital 2 is high. Please make sure you're up to date and try it again.

Please note that the scale switching expects a gate, and not a trigger. If you're using a gate, you have a voltage or hardware issue somewhere.

texasheadache, 2021-04-22:
Thank you for looking into this for me! I appreciate it. I am able to
switch to scale 2 when the gate at digital 2 is high - i just checked and
that works. I was under the impression that just by turning the knob to go
to the scale 2 page, then the scale being output would switch to scale 2.
Is this not accurate? Understanding this may help me recognize if my unit
is defective or if this is expected behavior.

Thanks for helping me out. I do think Hemispheres is great.

Chysn, 2021-04-22:
The only way to play scale 2 is for digital 2 to be high. It doesn't matter which scale you're currently editing.

texasheadache, 2021-04-22:
Ah - thank you so much for clearing that up for me! The confusion is all
mine. I appreciate your work on Hemispheres and for taking the time to help
me out.

closed 2021-05-01 by Chysn