Chysn/O_C-HemisphereSuite #87: Dual Quantizer Root/Transpose Not Working Correctly

jcoplin, 2021-04-30:
After setting the scale, for example to Dorian, changing the root from C to other roots does not transpose the CV. It just steps up a semi tone and then back and repeates as you go up the roots. For things that do not have a large fine tune control like the Atlantis, Model D, etc. this means the quantizer is unable to work in any other scale roots except C.

Chysn, 2021-05-01:
What do you mean by "it just steps up a semi tone and then back?"

Chysn, 2021-05-01:
Keep in mind that changing the root won't necessarily change the CV. Consider if

1) The selected scale is Dorian
2) The selected root is C
3) The CV note in is closest to C

then the quantizer will output a C. Now, if you change the root to D, the CV should remain the same. This is because the key of D Dorian contains C as its seventh degree. Root is not a kind of transpose at all; it just specifies the notes that are allowed to be output.

jcoplin, 2021-05-01:
I see. It seems though that it should as the root of the key also should conform to whatever "0v" is so that moving the root transposes steps correctly. Otherwise, it is impossible to use sequencers in a controllable state.

For example, with a 2v swing from off to full on means that off, middle, and on are all octaves and predictable. If the quantizer does not transpose, then those key positions will no longer correspond to the root and thus the quantifying will be in neither the correct mode nor key.

James

Airell, 2021-05-01:
I think you are referring to transposing. As I do would like that as an addition (but how?), the functionality of the root (actually the Tonic because we are talking about scales/modes, not chords) is a different functionality than transposing.

Although 'you go up' a whole note from C to D, one still needs to follow the Dorian (or other modes) rules of whole and half steps. It can be confusing. D Dorian follow all the white notes, just like the C Major (Ionian) mode. Only the root (Tonic) is different, so for a quantizer, D Dorian of C Major will not make any difference.

I use a precision adder to transpose.

Chysn, 2021-05-01:
@jcoplin, sorry, I’m not sure what you mean. But it seems like @Airell is on the right track in suggesting a precision adder. Since this is a bug report system, I’m going to close this issue. But feel free to hop out to the modwiggler topic (Hemisphere Suite or otherwise), where many people will happily discuss transposition strategies.

kind regards,
Jason

closed 2021-05-01 by Chysn