Chysn/O_C-HemisphereSuite #88: Feature Request: Transpose as part of Dual Quantizer

jcoplin, 2021-05-01:
If there is a better place for this, please let me know as the I now understand that the Dual Quantizer works as designed, but without an intgrated transpose, it is at odds with it's own design.

First, while it is true that without transpose, the resultant notes may be quantized into the correct pitch buckets, it does not mean that those same notes are in the correct scale/mode. Without transposing to accomidate the root, you are not in the listed mode. Certainly, C major (ionian), A minor (aeolian), D dorian, E phygerian, F lydian, G mixolydian, B locrian all share the same notes, namely C,D,E,F,G,A,B but to say that makes them the same is clearly not correct. A minor and C major are most definately not the same and it is because the root is different which makes for different phrasing and gravity of note cadence. 

If, you believe they are all the same, why have the modes available as settings at all? Under this reasoning, you can get all the modes in all the keys simply by having any single scale and using the root to approriately hit all correct pitches. In other words, if the quantizer had major as a scale option, you can get all other minor, dorian, lydian. etc. out of it by changing the root.

The other issue is that without transposition, how do you tune? If I set the Dual Quantizer to D dorian, with nothing plugged in I would expect to be able to tune an oscilator to D and incoming voltages should map to D dorian. However, this is not what happens. I assume (not sure what the default note value for 0 volts the Dual Quant uses) the no input voltage outputs C. So, If I tune my oscillator to D what results is D,E, F#, G, A, B, C#, D which is not D dorian but D ionian (major) which is not the same thing.

Sequencers typically output voltages in 1v, 2v, 4v, 8v type settings for a reason. It means that they generate 1, 2, 4, 8, etc. octaves over their ranges. If I have a sequencer set to 2v, without a transposition, I can not get it to track D dorian over a 2 octave range as I would have to start the root of the sequence not at 0V but at some increment up from there. Of course, an adder can be used to add an offset and make it correct but that is precisely why it should be in the quantizer. Without it, the quantizer is not really providing the selected mode and all the various scales and keys in the software are essentially pointless and you might well just have picked any of them and expect the user to do the work to figure which modes map to which.

Chysn, 2021-05-01:
Unfortunately, the Teensy's flash is full. There are no additional enhancements on the roadmap for Hemisphere Suite.

closed 2021-05-01 by Chysn

Chysn, 2021-05-01:
_If there is a better place for this, please let me know_

There are several modular forums where you can either find discussions about transposition or make a new one. I would recommend modwiggler.com or Lines (https://llllllll.co/).

Chysn, 2021-05-01:
_If I set the Dual Quantizer to D dorian, with nothing plugged in I would expect to be able to tune an oscilator to D and incoming voltages should map to D dorian_

This sounds like your primary misunderstanding. You should keep your oscillators tuned so that 0V outputs a C. This is the basic assumption of the quantizer.

jcoplin, 2021-05-01:
Then it isn't D Dorian. It's C Major. Fixing it really is  a matter of labels due to the relative nature of the modes. For example. If I set the quantizer to Dorian and the root C, if I tune the output to D, you get expected behavior. Same is true of all the other keys. Dorian C# on the o_C is E (if you tune the 0 input to E), and so on. Same is true of the other modes. You can set the o_C to A minor but what it outputs isn't A minor, it's C Major. If you are really stuck on this approach, you might as well as get code space back by removing the other modes as they are not at doing what they show.

Chysn, 2021-05-01:
_For example. If I set the quantizer to Dorian and the root C, if I tune the output to D, you get expected behavior - ie D Dorian._
 
Yes, that's the way to do it if you're going to retune the oscillator with every key change. But that's more work than it needs to be. You really should address this to a forum, as there are millions of people who can explain it better than I can.

Airell, 2021-05-02:
Sorry Chysn, as this is not the appropriate place to address this. For me the Dual Quantizer was one of the reasons to dig into chords/modes and it's a very interesting theory.

In Eurorack world, the rule is 0v is C and 1v/oct, just like an A is 440Hz spread over 12 notes. This is NOT the same as the Root (Tonic) of a Chord/Mode. 'You', the sequence, will play in the key of C or D etc... Just like the piano player. This means the _song_ will have 'a natural tendency' to have the C or D as 'starting or ending' note. The quantizer is merely to have oscillators in tune with it's setting, just like the piano is tuned.

Some one liners comments (sorry):
"If there is a better place for this, please let me know": ModWigglers.
"Then it isn't D Dorian. It's C Major.": for a (simple) quantizer, it's doesn't matter. It quantizes (rounds) an incoming voltage to the nearest key in that mode.
"why have the modes available as settings at all": It's a guideline. Set the quantizer to the mode your song is in and it will make you can't play anything 'out of the mode'.
"how do you tune", you don't. 0v in is 0v out, In D Dorian, 0v is accepted as a 'valid' note.
"If I tune my oscillator to D": In Eurorack, the rule is 0v is C. Have your sequencer play in the key of D. Just like the piano player plays in the key of D, instead of re-tuning the piano for every new song.
"If I have a sequencer set to 2v, without a transposition, I can not get it to track D dorian": Yes, you can not play the notes above '2v', but the quantizer is not for compensating that 'limitation of the chosen settings'. Your sequencer is now like a keyboard with a 2 octave range and now you want all the 'C' notes be tuned to 'D' because you can't play a full range 'D Dorian'? That's a bit strange.

I'm sorry, but I think you are trying to use the (simple) function of a quantizer to change the key of your sequence with a flick of a button. That's the sequencer/piano player's job, not the quantizer/piano tuner's.

Chysn, 2021-05-02:
@Airell Thank you, that should be helpful.

jcoplin, 2021-05-02:
This isn't a matter of me not understanding how a quantizer works. I get all that. This is fundamentally about the fact that what the quantizer is stating is incorrect. The fact that the same sequence input will sound exactly the same in C Major, D Dorian, E Phrygian, F Lydian,  G Mixolydian, A Aeolian, and B Locrian is absolutely wrong. The start of the D Dorian scale is D. Period. 100% of the time. The fact that all these modes are rooted on C means all of them are not representative of the mode except for Ionian (Major). This is incredibly misleading. It would be far more accurate to just have the  C Major and none of the other modes listed that share the same notes if you are not accounting for the root.

I also cannot see how 0v = C is so unbending when we are talking about a device whose sole purpose is to reassign pitches. Why is it OK to take the voltage defined for a G and remap it to a G# but not move C? If your argument is that it is up to the player, why have a quantizer at all? Additionally, the consistency of 0v = C comes from keyboard voltages. If we were talking about keyboards here, that woukd be expected. However, we aren't. 

I realize this isn't going to be changed. I've already pulled the code and fixed it for myself anyhow. If you were to do what I'm suggesting, try these modes with proper accommodation for the root, you would realize how important this is for anyone wanting to write in a mode or key to capture emotional content. 

There is a reason most sequencers have outputs that scale to 1v, 2v, 4v, 8v. Thus means that at least min, max, and middle knob positions all correspond to octaves. It also means this note appears more often than any other note on the knob range. This should be set to the root of the mode you are playing as this should be the most commonly played note in a melodic sequence. So, to actually play a sequence in D Dorian for example, a knob in the middle position, min or max would play D. If this isn't the case, you are leaving it to the performer to have to tune each step to be in mode plus you woukd lose however many notes are spaced between the root of your mode and C. 

Of course, traditional this is done with a fixed voltage source of some type and a precision adder. The advantage of digital implementations is that since this function is really essential and is "always" patched this way, it only seems correct that it should be in the digital quantier that displays roots.

Chysn, 2021-05-02:
There's an infinite number of ways to do modular. I'm glad you've found yours.