Chysn/O_C-HemisphereSuite #111: Compile fails if Enigma is swapped out for another app in OC_options.h

nickstokes, 2022-08-09:
When I swapped out ENABLE_APP_ENIGMA for another app in OC_options.h I get the compile error **'OC::Strings' has not been declared**

It looks like OC_strings.h is only included in APP_ENIGMA.ino, but is needed for OC_calibration.ino to compile.
Moving the line `#include "OC_strings.h"` from APP_ENIGMA.ino to o_c_REV.ino seems to fix it.

Chysn, 2022-08-09:
Thanks for the report! I’m not going to make this change to the codebase, since the final release is out and it doesn’t affect functionality. But anybody wishing to do swap-out of Enigma in the future can refer to this for help.

closed 2022-08-09 by Chysn

nickstokes, 2022-08-09:
This is actually an issue in the 'Benispheres' fork, and not this original version.
I accidentally submitted to the wrong repo. Whoops!