Chysn/O_C-HemisphereSuite #112: Dual Quantizer doesn't work in flipped firmware

risro, 2022-08-24:
It seems like the dual quantizer doesn't work with the flipped firmware. normal firmware works just fine, but if I use the flipped one it doesn't do anything!

Chysn, 2022-08-24:
I was able to confirm that both Scale Duet and Dual Quantizer work as expected with the Flip 180 version of 1.99Repeating. Double-check your patching, noting that Scale Duet has only one CV input and one CV output, and that it requires a clock input. Note that all three of these will use the left-hand jacks from the point of view of a flipped module.

closed 2022-08-24 by Chysn

risro, 2022-08-25:
sorry you are correct. I didn't realize that I needed to use the left handed jacks (I'm sure I tried that at some point, but now atleast it seams to work just fine?) thank you  a lot!