Chysn/O_C-HemisphereSuite #113: Feature request: Midi start/stop

kraftf, 2022-09-13:
Hi there quite new the hemispheres universe.
From what I can tell there is no way to have a reset pulse or gate from a midi start or stop message. Is it possible to implement one for captain midi and/or for the midi/cv applet?
Keep up the good work.
Cheers

Chysn, 2022-09-13:
It's a good idea, but Hemisphere Suite development is at its end.

closed 2022-09-13 by Chysn

kraftf, 2022-09-13:
Really? That's very sad! Wish I had given  this a shot earlier!!Anyway thanks alot.

Chysn, 2022-09-15:
There's just no space left to do anything, and also the only version of Arduino IDE that can build it no longer works on my computer.

synthi, 2022-11-09:
> Hi there quite new the hemispheres universe. From what I can tell there is no way to have a reset pulse or gate from a midi start or stop message. Is it possible to implement one for captain midi and/or for the midi/cv applet? Keep up the good work. Cheers

Maybe you can Ask Benirose to do it for the Benisphere branch??

Chysn, 2022-11-10:
Benirose made room for additional applets partly by removing all MIDI capabilities.

synthi, 2022-11-10:
He removed Captain MIDI app by default in lastest release but you can have
it back in options if you remove other apps. I have mine with Captain MIDI
and Darkest Timeline removing Enigma (but having Enigma Jr working). Also
MIDI in and MIDI out are there by default