Chysn/O_C-HemisphereSuite #116: Add TB3-PO App from Logarhythm1

cryptix, 2022-12-06:
This fork has a very nice sequencer: https://github.com/Logarhythm1/O_C-HemisphereSuite/blob/logarhythm-branch/software/o_c_REV/HEM_TB3PO.ino

I feel capable to do this from a coding standpoint but I'm having trouble to get the old Arduino/Teensy toolchain to work on my macbook...

Chysn, 2022-12-06:
Cannot move TB3PO to this codebase.

closed 2022-12-06 by Chysn

cryptix, 2022-12-06:
Understood, thanks.

Do you have a suggestion what Linux distro to use to get the toolchain running?

Chysn, 2022-12-06:
> Do you have a suggestion what Linux distro to use to get the toolchain
running?

I've always used macOS, but everything pretty much stopped working with
Montery.

You should be able to use any up-to-date Linux distro, as long as you can
run the 1.8.1 Arduino IDE.

--jason

cryptix, 2022-12-06:
fwiw, if somebody else finds this, I removed captain midi and pong over [here](https://github.com/cryptix/O_C-HemisphereSuite/tree/tb3po) which made enough space for it.

(the supplied image from logarithm doesn't work correctly with the VOR adjustments)