Chysn/O_C-HemisphereSuite #118: Impossible to upload

nicolaskern, 2023-01-25:
Hi,

I have managed to compile the code using Teensyduino, but when plugging the module, pressing the button, the teensy does not receive the code, and the IDE seems to be unable to detect my teensy.
What did I miss?

Thanks,
Nicolas

suit4, 2023-01-29:
Hi Nicolas,
sounds actually like a solvable problem. 
As long as the IDE does not see the teensy, you won't get anywhere.
Are you trying to do this on Windows, Mac or Linux?

And did you follow the procedure mentioned here: 
https://ornament-and-cri.me/firmware/ ?

nicolaskern, 2023-01-30:
Hi,

The problem was actually the cable... I tried many cables, but the only one working was an official Arduino cable!
It went fine uploading the HEX file.

Thanks!

closed 2023-01-30 by nicolaskern