Chysn/O_C-HemisphereSuite #122: Carpeggio Sequencing Unintuitive

snydoodlsh, 2023-08-12:
I'm using a Plum Audio O&C X. 1.8A.

As I use the Carpeggio app, a trigger sent to the trigger input advances the player to the next matrix square (position 1 to position 2), and the module plays the note from the first square. That doesn't seem like a big deal, except that I would expect this app to play the note of the matrix it lands on when it advances onto that square - not when it leaves the square it came from. This makes it difficult to know what note just played because it will play a "C" in the first square, but it instantaneously advances to the next square and shows you an "E". This also affects reseting the sequence. I'll reset the sequence with an input and nothing "plays". I would expect reseting actually plays the first "note" / first square. Maybe I'm weird for thinking this way. In it's current form, it makes it hard to use this app.

Chysn, 2023-08-12:
Please note that Hemisphere Suite is no longer in active development, and no further issues will be addressed nor new features implemented. See active forks for potential redress.

Chysn, 2023-08-12:
Check the latest version, 1.999…, for updates to reset and clock advance behavior across all applets that needed it. I don’t know if it’ll work with Plum Audio devices, though, but maybe worth a shot?

closed 2023-08-12 by Chysn