Chysn/O_C-HemisphereSuite #123: CV Attenuators on Plum Audio OC X mess up incoming CV for Dual Quantizer.

snydoodlsh, 2023-08-12:
I'm using a Plum Audio O&C X. 1.8A.

I've been trying to use incoming CV to be quantized using the Dual Quantizer app. My module has CV attenuators for incoming CV, but when I send it CV that is a "note value", it either attenuates it to "zero" and only a single note is quantized, or as I turn up the attenuators, it exponentially scales / attenuates the signal so that it can never scale to a normal "scale". If I send it simple sequence of notes from across 2 octaves starting with 0 volts (C3) I can't get a decent sounding sequence because the attenuverters either force me to suppress the signal entirely, or it scales it up disproportionately. I find that I have to take my note CV and put it into a dual attentuator / adder module to "shift" the notes up / down to compensate for the crude quantizer interpolation. It would be nice if I could turn off the attenuverters on my module or use them to "shift" notes up or down. 

https://www.plum-audio.com/product-page/ocp-x

Chysn, 2023-08-12:
Please note that Hemisphere Suite is no longer in active development, and no further issues will be addressed nor new features implemented. See active forks for potential redress. Specifically, I’m unable to support alternate hardware.

closed 2023-08-12 by Chysn

Chysn, 2023-08-14:
Fos Plum Audio device support, the Phazerville fork is active:

https://github.com/djphazer/O_C-BenisphereSuite/releases/tag/PSv1.5.1

snydoodlsh, 2023-08-14:
Thanks!

Matt Snyder

- - - - - - - - - - - - - - - - - - -
Medium <https://medium.com/lucid-software-design>
(801) 857-7476
- - - - - - - - - - - - - - - - - - -