Chysn/O_C-HemisphereSuite #126: Midi out not working

arpeggio1234, 2023-10-17:
Hallo, this firmware is great, i use it alot.
Only i can not get CV to midi working, not on midi-out and not on captain midi.
I connect the gate in to digital 1 in and the pitch CV to cv1 in. And then play my arturia minibrute.
I see CV comming in, but the notes they are all g9. And i see note ons and offs, all g9.
Any clou what am i doing wrong, what have i forget. I set the midi channel to 1.
Thanks in advance, Dian from the Netherlands

Chysn, 2023-10-20:
I'm unable to reproduce this issue with my Arturia MicroBrute using its CV and Gate outputs into MIDI Out.

MicroBrute's CV does run high, and MiniBrute is probably the same. Even the lowest octave setting, the lowest key is C4. So you're probably just pegging the high range. I'd suggest measuring the CV output with a meter. It may be necessary to run it through an attenuator.

I'm closing this issue, as it's probably more of a support issue than a software one.

closed 2023-10-20 by Chysn

arpeggio1234, 2023-10-25:
Hallo again.
Still running at the same issue.
Measured the voltages putted out by the o-ctrl and they are apx 1 2 or 3 volts and the captain-midi app is generating midi-notes but all g9.
Gate is good, it's going on and off.
So i am still reproducing this issue, where can i get support for this, thanks again. Dian

Chysn, 2023-10-25:
Probably a hardware issue on the module's ADC. There are no problems in the software.

arpeggio1234, 2023-10-26:
Yeah, thanks for the reply.
I use the After later Audio uO_C
What are the Analog input numbers 
for the CV ins and the digital input numbers fir the trigger inputs. So i can make a sketch myself and test if they work with a simple AnalogRead when DigitalRead is HIGH.
Thanks in advance, Dian

Chysn, 2023-10-27:
It sounds like you're asking about the hardware connections, and I'm afraid I don't know anything about that.

arpeggio1234, 2023-10-27:
These pins are hardware connection, but you wrote the code so which pin numbers are used in the code for cv1 cv2 cv3 and cv4 and for trigger input 1 2 3 and 4.
I mean when you write a arduino sketch for the teensy how do you name cv1 etc.
Thanks again, Dian

Chysn, 2023-10-27:
I didn't write that part. But the source code is RIGHT HERE if you'd like to review it.

arpeggio1234, 2023-11-17:
Completely solved this.
After several tryes with the uO_C from After Later Audio i found that when i use modular case power it reacted good so scope give good results on cv in. 
But when i swiched to powen over usb then the scope cv ins did not work anymore.
So there is a, hardware related, point i think.
Maybe cause usb power is only 5 volt?
So anyway, i cutted the usb piwer trace on the teensy 3.2 board and now i can use modular case power and the usb cable for the midi out app. Works great. Solved!

Chysn, 2023-11-17:
Excellent, great to hear!